evert-arias/EasyBuzzer #1: Buzzer beeping forever

capedra, 2018-03-03:
Hello, I've tested the SingleBeep sketch on Lolin32 and the buzzer keeps playing forever. I already tested GPIOs 32,33,27,2,4,etc but the problem persists on every pin.
The workaround for me is to add ```EasyBuzzer.stopBeep();``` on doneBeeping function. It only works this way.

Thanks!

capedra, 2018-03-07:
@evert-arias Oh, and congratulations for this very simple library! Great job! Thanks!
So, can you fix this issue?
Also, in one of my projects, I use more than one buzzer at the same time, so, it would be nice if you had some examples which included ```EasyBuzzerClass buzzer1, buzzer2; buzzer1.setPin(4); buzzer2.setPin(5); etc...```, because maybe it could help people out. Just wondering. See you!

capedra, 2018-03-07:
@evert-arias 
One more thing for the function that will be called when the beeping sequence ends:
```
int pin = 2;
void finishedBeeping() {
buzzer1.stopBeep();
pinMode(pin, INPUT);
Serial.println(F("Finished beeping!"));
}
```
This pinMode will make the buzzer come back to its original state (INPUT mode), making its noise disappear and helping people out.

To be honest, let's make everything simple:
```
int pin = 2;
void finishedBeeping()
{
  pinMode(21, INPUT);
  Serial.println("Temperature from ESP32: "+String(temperatureRead())+"C/"+String((temperatureRead() * 9.0) / 5.0 + 32)+"F");
}
```

evert-arias, 2018-03-07:
@capedra 
Hello, I think the workaround you've used is very well formulated and it should work. However, at some time this week, I will look for a practical solution for this issue. I will also add support for more than one instance of `EasyBuzzerClass `because for the moment it only allows a single instance `EasyBuzzer` that can be accessed without having to create it to make it easier to use the library. But as you recommend, it would be good to have the possibility of several instances. Feel free to report any other issue that may appear, I will do my best to solve it as soon as possible. I appreciate your collaboration.

evert-arias, 2018-03-09:
I've tested the example in an ESP32 and it did not turn out to have the issue you mentioned. It would be good if you shared the sketch you used for the test that gave you the error.

osos, 2018-03-14:
I have seen the same issue on my own buzzer-code based on Ticker events. (https://github.com/bertmelis/Ticker-esp32)

the pinMode(pin, INPUT) trick solved the issue.

Thank you @capedra

capedra, 2018-04-25:
@osos You're welcome. I just keep having this same issue while beeping on setup function only. It works fine on loop. So, I wrote some code for the buzzer to play on the 2nd core of the ESP32. It's very simple. I will write it here asap.

dalidavila, 2018-08-11:
Hi, 
I don't know if I'm using the library in a wrong way but theres is some issues I had so far.

1- function min and max wasnt recognized, so I had to edit EasyBuzzer.ccp on lines where those are used, seems to be that the old versions of de Arduino IDE used another implementation of those functions, I readed something like that on forum but I lost the link. All you have to do is to cast the arguments of the min and max function when you call them to make the parameters have the same type.

2- I'm running my sketch on a Wemos D1 R2 mini and  __AVR_ATmega328P__ and ESP32 doesnt get defined, so I edited the  EasyBuzzerClass::update() like this and now it sounds:
`#if defined __AVR_ATmega328P__ // ARDUINO
	if (timeInSequence < blinkingDuration && timeInSequence % blinkDuration < mOnDuration) {
		Serial.println(String("DEBUG: TONE"));
		tone(mPin, mFreq);
	}
	else {
		Serial.println(String("DEBUG: noTONE"));
		noTone(mPin);
	};
	return;
#endif
#if defined ESP32 // ESP
	if (timeInSequence < blinkingDuration && timeInSequence % blinkDuration < mOnDuration) {
		Serial.println(String("DEBUG: ESP32 - TONE"));
		ledcAttachPin(mPin, mChannel);
		ledcWriteTone(mChannel, mFreq);
	}
	else {
		Serial.println(String("DEBUG: ESP32 - noTONE"));
		ledcDetachPin(mPin);
	};
	return;
#endif
	// OTHERS
	if (timeInSequence < blinkingDuration && timeInSequence % blinkDuration < mOnDuration) {
		Serial.println(String("DEBUG: OTHERS - TONE"));
		tone(mPin, mFreq);
	}
	else {
		Serial.println(String("DEBUG: OTHERS - noTONE"));
		noTone(mPin);
	};`
3- I edited this function just be sure that the pin is correctly initilyzed and started "off"
`void EasyBuzzerClass::setPin(unsigned int pin) {
	mPin = pin;
	pinMode(mPin, OUTPUT);
	digitalWrite(mPin, LOW);
}`

4- the buzzer nevers stops, my sketch is dummy one:
`#include <Arduino.h>
#include <EasyBuzzer.h>
int speakerPin = D8;
void setup() {
	Serial.begin(9600);	 // Initialize serial communications with PC
	while (!Serial) { ;  }
	EasyBuzzer.setPin(speakerPin);
}
void loop() {
EasyBuzzer.singleBeep(
	  2500,	// Frequency in hertz(HZ).
	  500	// Duration of the beep in milliseconds(ms).
);
delay(5000);
}`

evert-arias, 2018-08-13:
Hi,
Thanks for the fixes done; **1**, **2** and **3**.

**4**- Can not use `delay()`, this will stop the normal behavior of the library.
Refer to: https://playground.arduino.cc/Code/AvoidDelay

I have been very busy working on other projects, so I appreciate any contribution to this library.
I would like to offer you the possibility of collaborating with the development of this library. I will be sending you a collaborator invitation.

shoteff, 2018-11-03:
Adding mStartTime = 0; in stopBeep() method fixes this problem. 
Otherwise when a continuous beep is started, it depends on the timing calculations, but the method stopBeep() doesn't change the conditions, hence the beep doesn't stop

evert-arias, 2018-11-04:
@shoteff I have merged your pull-request. Please, feel free to continue providing feedback if you found anything that could help this library. Thank you for your collaboration. Best regards

evert-arias, 2019-02-15:
This issue is consider to be solved

closed 2019-02-15 by evert-arias

proddy, 2019-02-25:
FYI I had same problem with the library on an ESP32. The only way to stop the buzzer is by calling `pinMode(BUZZER_PIN, INPUT)` after the `ledcDetachPin(BUZZER_PIN)`.