evert-arias/EasyBuzzer #2: Problem Building any sketch using EasyBuzzer with Arduino 1.8.5

Pelado-Mat, 2018-05-03:
`Arduino:1.8.5 (Windows 10), Tarjeta:"Arduino Nano, ATmega328P (Old Bootloader)"

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp: In constructor 'EasyBuzzerClass::EasyBuzzerClass()':

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp:14:40: error: 'ledcSetup' was not declared in this scope

  ledcSetup(mChannel, mFreq, mResolution);

                                        ^

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp: In member function 'void EasyBuzzerClass::stopBeep()':

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp:66:20: error: 'ledcDetachPin' was not declared in this scope

  ledcDetachPin(mPin);

                    ^

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp: In member function 'void EasyBuzzerClass::update()':

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp:104:31: error: 'ledcAttachPin' was not declared in this scope

   ledcAttachPin(mPin, mChannel);

                               ^

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp:105:32: error: 'ledcWriteTone' was not declared in this scope

   ledcWriteTone(mChannel, mFreq);

                                ^

C:\Users\Matias\Documents\src\home_sensors\libraries\EasyBuzzer\src\EasyBuzzer.cpp:108:21: error: 'ledcDetachPin' was not declared in this scope

   ledcDetachPin(mPin);

                     ^

exit status 1
Error compilando para la tarjeta Arduino Nano.

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

`

per1234, 2018-05-04:
It looks like you're using the 1.0.0 release of the library, which only is compatible with ESP32, but you're compiling for Arduino Nano. Support for the Nano has been added since the 1.0.0 release (https://github.com/evert-arias/EasyBuzzer/commit/45010f943b686f3059c897ada19a483761a08cb2) so you would need to update to the beta version of the library or else wait for the next release to use it with your Nano.

Pelado-Mat, 2018-05-04:
@per1234 I used the one thats available in Arduino:1.8.5. I wil try installing manually the lib.

Thanks!

closed 2018-08-05 by evert-arias