evert-arias/EasyBuzzer #5: [Request BugFix] error: no matching function for call to 'max(int, const unsigned int&)'

LuckyLuzz, 2019-01-23:
Hey,
recently the ESP32 Core for Arduino has been updated and now it is not possible to Build the Library.

The compiler error message is:

```
C:\Users\XXX.platformio\lib\EasyBuzzer_ID1883\src\EasyBuzzer.cpp: In member function 'void EasyBuzzerClass::beep(unsigned int, unsigned int, unsigned int, byte, unsigned int, unsigned int)':
C:\Users\Thomas\.platformio\lib\EasyBuzzer_ID1883\src\EasyBuzzer.cpp:33:61: error: no matching function for call to 'max(int, const unsigned int&)'
mOnDuration = onDuration ? max(MINIMUM_INTERVAL, onDuration) : 0;
^
```

Environment: 
Win10, VS Code (V1.30.2)
PlattformIO (V3.6.3) 
Plattform Espressif 32(V1.6.0) with ESP-IDF 3.1.2

evert-arias, 2019-01-24:
@shoteff Hi,
I already found a solution to this issue and I will be updating the code as soon as possible. Thanks for reporting the issue.

evert-arias, 2019-02-15:
This issue is consider to be solved

closed 2019-02-15 by evert-arias

pathnirvana, 2023-02-10:
there are still many missing max function errors in other boards. I am using a stm32 board

pathnirvana, 2023-02-10:
fixed by adding `#define MAX(X, Y) (((X) < (Y)) ? (Y) : (X))`