evert-arias/EasyBuzzer #9: dangerous defines

proddy, 2019-02-19:
not a bug as such, but be careful with the #DEFINES in EasyBuzzer.h

```
#ifdef ESP8266
#define min _min
#define max _max
#endif
#ifdef ESP32
#define min _min
#define max _max
#endif
```

(regardless that they are the same) Any other code that uses the words **min** or **max** will be substituted at compile time.

Mas7ro, 2019-10-17:
ok, maybe is other kind of error:
`Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':
(.text+0x0): multiple definition of `__vector_13'
/var/folders/_6/21prk2wd4dj9btzv4287bzsm0000gn/T/arduino_build_739363/libraries/IRremote/IRremote.cpp.o (symbol from plugin):(.text+0x0): first defined here
/Users/Mastro/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
collect2: error: ld returned 1 exit status`

Mas7ro, 2019-10-17:
It was a conflict with IRremote, this (https://forum.arduino.cc/index.php?topic=120955.0) solved for me.

dmenne, 2019-12-19:
I agree with @proddy; these redefinitions cause a lot of problems with other code.

VTeselkin, 2021-12-02:
Delete `#ifdef ESP8266
#define min _min
#define max _max
#endif
#ifdef ESP32
#define min _min
#define max _max
#endif`
in EasyBuzzer.h
Change to: `/* Beep sequence at a given frequency. */
void EasyBuzzerClass::beep(unsigned int frequency, unsigned int const onDuration, unsigned int const offDuration, byte const beeps, unsigned int const pauseDuration, unsigned int const sequences)
{
	mFreq = frequency;
	mOnDuration = onDuration ? _max(MINIMUM_INTERVAL, onDuration) : 0;
	mOffDuration = offDuration ? _max(MINIMUM_INTERVAL, offDuration) : 0;
	mBeeps = beeps;
	mPauseDuration = pauseDuration ? _max(MINIMUM_INTERVAL, pauseDuration) : 0;
	mSequences = sequences;
	mFinishedCallbackFunction = NULL;
	mStartTime = _max(millis(), 1);
	mLastRunTime = 0;
	update();
}
/* Beep sequence at a given frequency, with callback functionality. */
void EasyBuzzerClass::beep(unsigned int frequency, unsigned int const onDuration, unsigned int const offDuration, byte const beeps, unsigned int const pauseDuration, unsigned int const sequences, void (*finishedCallbackFunction)())
{
	mFreq = frequency;
	mOnDuration = onDuration ? _max(MINIMUM_INTERVAL, onDuration) : 0;
	mOffDuration = offDuration ? _max(MINIMUM_INTERVAL, offDuration) : 0;
	mBeeps = beeps;
	mPauseDuration = pauseDuration ? _max(MINIMUM_INTERVAL, pauseDuration) : 0;
	mSequences = sequences;
	mFinishedCallbackFunction = finishedCallbackFunction;
	mStartTime = _max(millis(), 1);
	mLastRunTime = 0;
	update();
}` 
in EasyBuzzer.cpp

mrNo0b, 2022-11-21:
I think there is also conflict with `#include <unordered_set>`, if I include unordered_set after EasyBuzzer it will not compile with error code 1, I use ESP32 DOIT Devkit V1