evert-arias/EasyBuzzer #10: Sound does not stop if update() is delayed

mvackel, 2019-02-21:
If the update() function is delayed, for instance because of a long process, the sound does not stop.
    
Suggestion: 
Change, from line 122:

    	if (!sequenceDuration || (mSequences != 0 && elapsedTime / sequenceDuration >= mSequences))
    	{
    		mStartTime = 0;
    		if (mFinishedCallbackFunction)
    		{
    			mFinishedCallbackFunction();
    		}
    		return;
    	};

To:

    	if (!sequenceDuration || (mSequences != 0 && elapsedTime / sequenceDuration >= mSequences))
    	{
    		mStartTime = 0;
    		if (mFinishedCallbackFunction)
    		{
    			mFinishedCallbackFunction();
    		}
    #if defined ESP32
    		ledcDetachPin(mPin);
    #else
    		noTone(mPin);
    #endif
    		return;
    	};

sixtyfive, 2020-08-16:
Spent two hours trying to debug this until I had the idea to come and check for an existing Issue. I guess the fact that it's more than a year old means that the project is abandoned?

evert-arias, 2020-08-16:
Hi @sixtyfive 
I am not actively working on this repository but in the next few days I will be solving this particular issue.

sixtyfive, 2020-08-16:
Thank you for the quick response, Evert, and sorry for the hopefully not too brazen assumption!

evert-arias, 2020-08-17:
> Thank you for the quick response, Evert, and sorry for the hopefully not too brazen assumption!

Not at all 👍

iqfareez, 2022-04-12:
Hi hii. So have any updates on this issue?